DSchroer/dslcad #2: .obj exporter?

MrLeap, 2023-02-10:
Would be handy to be able to export the meshes as something commonly supported by gamedev engines. 

.obj seems suitable to the task, it's a very easy, human readable format. The only format perhaps more ubiquitous is .fbx, which is far less friendly.

.gltf is another good candiate.

Really any of these would open a gateway of silliness.

virtualritz, 2023-02-10:
It would be even more handy if the OBJ could also/alternatively contain the (trimmed) NURB surfaces from the b-rep.

Few people today know this but Waveront OBJ is [perfectly capable of storing these](http://paulbourke.net/dataformats/obj/) (scroll down to **Free-form curve and surface types**).

Among the apps that ingest such OBJs with higher order geometry are Maya and Rhinoceros 3D.
The latter together with  the procedural Grashopper modeling tool makes for a *very* interesting combo with DSLCAD.

I wanted to look into adding at least NURBS + trim curves support to [`tobj`](https://crates.io/crates/tobj) anyway.

Generating the `.obj` data through Maya &| Rhino was the plan but if this comes to DSLCAD I'd prefer that to get some generative regression testing going for NURBS support in `tobj`. I.e. DSDLCAD→`.obj`→`tobj`.

DSchroer, 2023-12-05:
I have moved the export format from `stl` which is basic to `3mf` to try to be forward looking. I think its better to support one format well rather than have a bunch of different formats. Its should be trivial to convert 3mf to obj with a tool like meshlabs. 

Closing this for now since multi type export is not something I think the project needs. There are plenty of good converter tools.

virtualritz, 2023-12-05:
The point I made in this ticket, was having a way to export NURBS that is easier to implement than STEP.

If you have a B-rep modeling tool, you want to have a way to export data as lossless as possible.

For example, if I want to create CAM tool paths in something like Fusion360, your input data should not be a triangle mesh.

Furthermore, if you mesh a surface, you can create a quad mesh and only use triangles sparsely, e.g. at trimmed edges.

AFAIK 3MF only support triangles.

The whole point of using a B-rep modeler is moot then if all you can export are triangle meshes. Use a SDF modeler instead. ;)

DSchroer, 2023-12-08:
The 3MF format is forward looking and moving to allow volumetric specification. At the moment its likely not very supported but I am not sure if OBJ B-rep is very supported either at this point.

https://github.com/3MFConsortium/spec_volumetric/blob/master/3MF%20Volumetric%20Extension.md

DSchroer, 2023-12-08:
What matters most if the development flow. User experience if you prefer that term. B-rep vs SDF should be hidden from the user in a tool like this. It shouldnt matter.

virtualritz, 2023-12-09:
I think the misunderstanding comes from the fact that you assume people export from `dslcad` and then just 3D print.

DSchroer, 2024-01-08:
@virtualritz from your experience. Would the standard BREP object file from Opencascade work or is it pretty non standard?

virtualritz, 2024-01-08:
STEP is kind of the industry standard for exchange of B-Rep data.  It's to B-Rep data what OBJ (still) is to mesh data.
OpenCascade is product-specific. Just like CATIA or Solidworks formats are. I.e. I wouldn't make supporting them a priority.

As for STEP: hence my suggestion to look at the `truck` crate, specifically their [`truck-stepio`](https://github.com/ricosjp/truck/tree/master/truck-stepio) which in turn is based on [`ruststep`](https://crates.io/crates/ruststep).

That said, I would still support OBJ, both for meshed geometry (especially if the mesher can produce n-gons instead of dicing everything into triangles) and for trimmed NURBS export.

Simply because it is a vety low hanging fruit and the NURBS part is easy to verify via Maya or Rhino3D demo versions. They both read OBJ files containing higher order geometry.

DSchroer, 2024-01-10:
Looks like I am in the position where work could start on this. 
From what I can tell there are no out of the box obj file libraries with free form curve support so it would need to be hand written.

@virtualritz would you be able/willing to help with that development?

DSchroer, 2024-01-10:
I suspect we could implement it here: `crates/persistence/src/lib.rs`

virtualritz, 2024-01-11:
> I suspect we could implement it here: crates/persistence/src/lib.r

Cool! But maybe change the name of this crate so you can eventuall publish it on `crates.io` (name clash, AFAICS).

Format specs are in the section [**Free-form curve/surface attributes** here](https://paulbourke.net/dataformats/obj/).

I can offer help like so:

### Sample Files

These can be inspected to see how a (reasonably) well-formed OBJ containing these geometry types should look like. 

* [ ] Contact a friend who's a professional designer and ask them to export OBJs from Rhino3D
  * A sphere.
  * Some simple CSG operation that results in a bunch of trimmed, joined surfaces.
* [ ] Ask another friend who's a VFX artist do the same from Maya.

I can contact different people for samples if you want to add STEP support, eventually.

### Testing

* [ ] Install a Rhino3D & Maya demo on my Windows VM and import OBJ files from `dslcad`.

## Code Review

I can also look at code but I can't make any promises when or how much.

I just started a new job and as such my spare time has become a scarce resource.

DSchroer, 2024-01-12:
Name may change later. Right now its not really intended for publishing. All the crates are pretty connected to dslcad themself.

virtualritz, 2024-02-23:
Here are the promised [examples from Rhino](https://github.com/DSchroer/dslcad/files/14389238/obj_examples.zip).

A sphere and some CSG union operation between a sphere and a cube.

DSchroer, 2026-09-21:
Added STEP export which can be move to other programs to convert. OBJ seems a bit out of scope right now.

closed 2026-09-21 by DSchroer