DSchroer/dslcad #4: UI scale factor not used on Kubuntu/KDE

virtualritz, 2023-02-10:
I wager this is coming from `bevy` as when I run a 'normal' `egui`/`eframe` app, the UI is scaled correctly. 

But in case of DSLCAD it uses 1.0 (my system uses 1.5).
When DSLCAD starts, `winit` says the usual (correct):
```
2023-02-10T18:04:19.605215Z  INFO winit::platform_impl::platform::x11::window: Guessed window scale factor: 1.5 
```

But everything is tiny, i.e. the actual UI definitely uses 1.0, not 1.5.

DSchroer, 2023-02-10:
Interesting. Ill try it out on my machine and see if I can get the UI to scale correctly. If it doesn't ill be sure to push a fix. Thank you for the bug report.

mjecke, 2023-02-10:
Just wanted to add that something like this seems to also be true on Mac.  The UI titlebar upon launch is tiny tiny tiny on a retina display.   Here is what is reported upon launch for my system:

`2023-02-10T22:02:19.479160Z  INFO bevy_render::renderer: AdapterInfo { name: "Apple M1 Pro", vendor: 0, device: 0, device_type: IntegratedGpu, driver: "", driver_info: "", backend: Metal }`

DSchroer, 2023-02-13:
Hey. I have pushed a new pre-release that should fix the UI scaling issues. Can you give it a try? 

https://github.com/DSchroer/dslcad/releases/tag/v0.0.2-pre.1

mjecke, 2023-02-13:
That indeed fixed the UI scaling on my Mac.   Toolbar looks correctly sized now
[image]

DSchroer, 2023-02-13:
Closing as resolved.

closed 2023-02-13 by DSchroer