DSchroer/dslcad #5: Crash to desktop with specific file

benthillerkus, 2023-02-11:
```ds
var column = circle(radius=5.0, center=true) -> shape extrude(z=10) -> shape fillet(radius=0.2);
difference(left=column, right=(column -> shape translate(x=1)));
```

benthillerkus, 2023-02-11:
x64 Windows 10 with an NVidia GPU

DSchroer, 2023-02-11:
So I have successfully reproduced the issue. It seems to be windows specific but also occurs under wine. Thank you for the report, I will try to get a fix out soon.

DSchroer, 2023-02-11:
Interesting enough building a windows version locally results in it working. Seems to be something broken with the CI setup that builds the pre-built versions.

DSchroer, 2023-02-11:
> Tbh I would have expected the culprit to be right=(...) being syntax I made up

That syntax seems fine. You can omit the `()` around it since it gets evaluated as an single argument already.

stolksdorf, 2023-02-26:
Also getting crashes right to desktop with some files, can't seem to figure out the culprit. (x64 Windows 10 with an NVidia GPU.)

Example `gear.ds` -> Crash
Example `fidget_cube.ds` -> Works
Example `aerogrind_funnel.ds` -> Works
Example `gopro_adapter.ds` -> Crash

`cube(x=10, y=10, z=10)->shape fillet(radius=10);` -> Crash
`cube(x=10, y=10, z=10)->shape fillet(radius=9);` -> Works

DSchroer, 2023-03-09:
@benthillerkus I did some work on the windows build pipeline. Can you try your sample on the latest pre-release? https://github.com/DSchroer/dslcad/releases/tag/v0.0.2-pre.5

On my end all of the examples work now.

DSchroer, 2023-03-09:
You will need to adjust the code slightly. 
```
circle(radius=5.0, center=true)
```
should now be:
```
circle(radius=5.0) ->shape center()
```

closed 2023-03-09 by DSchroer