DSchroer/dslcad #6: Allow writing out text files --- this would allow G-code generation for CAM

WillAdams, 2023-02-11:
I've been working on a tool which allows one to model how a CNC cuts using OpenSCAD:

https://github.com/WillAdams/gcodepreview

which sort of works using RapCAD (have actually done a couple of test projects), but it is difficult matching the state-based nature of G-code controlling a machine in OpenSCAD (maybe someone who knew that language better could resolve this), and the need to use an obscure tool (RapCAD) is a bit awkward.

It would be nice to have direct control over writing out text files in the programming tool, and to have a language which would allow one to note machine state after a module/function call and then use that later.

DSchroer, 2023-02-11:
Right now if you render parts that are text they print out to the console. Would it be possible to build upon that to achieve what you want? 

There isn't a ton of tooling around text editing yet. I would be happy to expand the offering if you have specifics in mind.

WillAdams, 2023-02-12:
For example, one of my modules is defined as:

    module rapid(bx, by, bz, ex, ey, ez) {
    if (generategcode == true) {
    	writeln("(rapid)");
    	writeln("G0 X",ex," Y", ey, "Z", ez);
    }
    hull(){
        translate([bx, by, bz]){
          select_tool(102);
        }
        translate([ex, ey, ez]){
          select_tool(102);
        }
      }
    }

It doesn't really do anything in the model (I should change the colour used for subtracting this to maybe function as a warning), but it needs to be able to only write out the G-code is specified to (it's rather a performance hit otherwise), and it needs to be able to write out hard-coded text, and convert variables to their string representation and write them out as well.

DSchroer, 2023-02-13:
How would this work if you have a subtractive operation? 
For a OpenSCAD example:
```
difference() {
  item1()
  item2()
}
```
How does that kind of operation affect the code generation?

WillAdams, 2023-02-13:
CAM on a traditional CNC is inherently subtractive.

If you look at the sample file, the code is

    difference() {
        setupupstock (<dimensions, position, origin, &c.>);
        gcut and other commands, all of which are subtracted from the stock
    }

see: https://github.com/WillAdams/gcodepreview/blob/main/gcode_flatten.rcad

DSchroer, 2023-02-13:
Got it. So the stock is always a solid and you generate a side channel of gcode. 

I think to start I'll do 2 things:
1. Allow 1D parts to be written out to text files. 
2. Add some utils for writing and editing text

With that you would be able to manipulate the GCode as if it were a normal part.

WillAdams, 2023-02-13:
Sounds like a plan! I'll keep an eye out for this and try porting my code over.

DSchroer, 2023-02-20:
Hey @WillAdams. I have added this feature in the latest pre-release. Can you try it out and see if that fits your needs. 

There are also some string utils `format` and `formatln` that will help you build your text. 

https://github.com/DSchroer/dslcad/releases/tag/v0.0.2-pre.3

closed 2023-02-20 by DSchroer