DSchroer/dslcad #9: Unable to find a GPU! Make sure you have installed required driver

setop, 2023-02-12:
What are the system requirements ?

```
$ RUST_BACKTRACE=full ./dslcad 
2023-02-12T12:08:56.889380Z  INFO winit::platform_impl::platform::x11::window: Guessed window scale factor: 1    
thread 'main' panicked at 'Unable to find a GPU! Make sure you have installed required drivers! For extra information, see: https://github.com/bevyengine/bevy/blob/latest/docs/linux_dependencies.md', /root/.cargo/registry/src/github.com-1ecc6299db9ec823/bevy_render-0.9.0/src/renderer/mod.rs:121:10
stack backtrace:
   0:     0x563a658ded2a - std::backtrace_rs::backtrace::libunwind::trace::h34aec3ef6cd8ad7e
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/../../backtrace/src/backtrace/libunwind.rs:93:5
   1:     0x563a658ded2a - std::backtrace_rs::backtrace::trace_unsynchronized::h8035d38698d0f7a8
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x563a658ded2a - std::sys_common::backtrace::_print_fmt::hff968f6695a1ba22
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/sys_common/backtrace.rs:65:5
   3:     0x563a658ded2a - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h7eea0efe77acf1ec
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/sys_common/backtrace.rs:44:22
   4:     0x563a6590739e - core::fmt::write::hc553f17407eb0b48
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/core/src/fmt/mod.rs:1208:17
   5:     0x563a658d9f55 - std::io::Write::write_fmt::h62e5f01a089f48c0
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/io/mod.rs:1682:15
   6:     0x563a658deaf5 - std::sys_common::backtrace::_print::h52d116aff3db4cd1
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/sys_common/backtrace.rs:47:5
   7:     0x563a658deaf5 - std::sys_common::backtrace::print::h9e7d2f98fb7af075
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/sys_common/backtrace.rs:34:9
   8:     0x563a658e03ff - std::panicking::default_hook::{{closure}}::hf74999dab2d0a95c
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/panicking.rs:267:22
   9:     0x563a658e013b - std::panicking::default_hook::hc11ca7d10c44c42f
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/panicking.rs:286:9
  10:     0x563a658e0b0c - std::panicking::rust_panic_with_hook::hdb4da1ae79c845a5
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/panicking.rs:688:13
  11:     0x563a658e08a9 - std::panicking::begin_panic_handler::{{closure}}::h02b5b35b126d5cf2
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/panicking.rs:579:13
  12:     0x563a658df1dc - std::sys_common::backtrace::__rust_end_short_backtrace::h6c6853376cf416d1
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/sys_common/backtrace.rs:137:18
  13:     0x563a658e05b2 - rust_begin_unwind
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/panicking.rs:575:5
  14:     0x563a63c3d343 - core::panicking::panic_fmt::hfd9e949092070b66
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/core/src/panicking.rs:64:14
  15:     0x563a65904ae1 - core::panicking::panic_display::h45b76cd63e812370
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/core/src/panicking.rs:135:5
  16:     0x563a65904a8b - core::panicking::panic_str::he2d190e15dd7e6c1
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/core/src/panicking.rs:119:5
  17:     0x563a63c3d306 - core::option::expect_failed::h240c1b84e7fce7a2
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/core/src/option.rs:1879:5
  18:     0x563a64643850 - bevy_render::renderer::initialize_renderer::{{closure}}::h59960cd9e1f3a947
  19:     0x563a646350cb - std::thread::local::LocalKey<T>::with::h71274ee9c519bf7a
  20:     0x563a646fc97b - <bevy_render::RenderPlugin as bevy_app::plugin::Plugin>::build::hf0a5dc2f01793b2c
  21:     0x563a64d424ed - bevy_app::app::App::add_boxed_plugin::h11d0a59f2d91c623
  22:     0x563a64d46066 - bevy_app::plugin_group::PluginGroupBuilder::finish::h1a9feacc1dfc859f
  23:     0x563a63cc0f1d - dslcad::editor::main::he65d9fbd04faa0c9
  24:     0x563a63c57a69 - dslcad::main::hce5bdefa6f9cc6f4
  25:     0x563a63c60a03 - std::sys_common::backtrace::__rust_begin_short_backtrace::h53532dbd9feb7c16
  26:     0x563a63c79939 - std::rt::lang_start::{{closure}}::h410593887c6bbf5f
  27:     0x563a658d2dec - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once::h9ab31282e87f134a
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/core/src/ops/function.rs:606:13
  28:     0x563a658d2dec - std::panicking::try::do_call::h42ddf5b01d0b4bc7
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/panicking.rs:483:40
  29:     0x563a658d2dec - std::panicking::try::hfb70320d7386c61a
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/panicking.rs:447:19
  30:     0x563a658d2dec - std::panic::catch_unwind::h978c9edbad2bb4d4
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/panic.rs:137:14
  31:     0x563a658d2dec - std::rt::lang_start_internal::{{closure}}::h04ede5bd2f26b553
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/rt.rs:148:48
  32:     0x563a658d2dec - std::panicking::try::do_call::ha6b9da35a0885c93
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/panicking.rs:483:40
  33:     0x563a658d2dec - std::panicking::try::h3325520cab3a642e
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/panicking.rs:447:19
  34:     0x563a658d2dec - std::panic::catch_unwind::h160beec6f047175b
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/panic.rs:137:14
  35:     0x563a658d2dec - std::rt::lang_start_internal::h79190e3a877a769d
                               at /rustc/fc594f15669680fa70d255faec3ca3fb507c3405/library/std/src/rt.rs:148:20
  36:     0x563a63c57ab5 - main
  37:     0x7f955a054d0a - __libc_start_main
                               at ./csu/../csu/libc-start.c:308:16
  38:     0x563a63c4db6a - _start
  39:                0x0 - <unknown>
```

DSchroer, 2023-02-13:
Right now I am using the defaults from `bevy` to handle the rendering. You should be able to run the CLI on almost any system. 

The bevy dependencies are here: https://github.com/bevyengine/bevy/blob/latest/docs/linux_dependencies.md

For your case I would say you need some kind of vulkan enabled driver. One of `vulkan-radeon`, `vulkan-intel`, or `mesa-vulkan-drivers`.

DSchroer, 2023-02-13:
I have updated the documentation to list the specific requirements. Ill close the ticket for now unless you have more specific issues regarding missing dependencies.

Symbian9, 2023-02-15:
I has NVIDIA GPU with `nvidia-legacy-390xx` drivers and `mesa-vulkan-drivers` installed, but it still fails:

```
$ dslcad
2023-02-15T16:10:30.214117Z  INFO winit::platform_impl::platform::x11::window: Guessed window scale factor: 1.1458333333333333    
thread 'main' panicked at 'Unable to find a GPU! Make sure you have installed required drivers! For extra information, see: https://github.com/bevyengine/bevy/blob/latest/docs/linux_dependencies.md', /root/.cargo/registry/src/github.com-1ecc6299db9ec823/bevy_render-0.9.0/src/renderer/mod.rs:121:10
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

DSchroer, 2023-02-15:
What do you get if you run `vulkaninfo` on your system?

Symbian9, 2023-02-15:
> What do you get if you run `vulkaninfo` on your system?

Here it is:

<details><summary>vulkaninfo</summary>

```
$ vulkaninfo
WARNING: [Loader Message] Code 0 : loader_scanned_icd_add: Driver libGLX_nvidia.so.0 supports Vulkan 1.3, but only supports loader interface version 4. Interface version 5 or newer required to support this version of Vulkan (Policy #LDP_DRIVER_7)
WARNING: lavapipe is not a conformant vulkan implementation, testing use only.
==========
VULKANINFO
==========

Vulkan Instance Version: 1.3.216

Instance Extensions: count = 20
===============================
	VK_EXT_acquire_drm_display             : extension revision 1
	VK_EXT_acquire_xlib_display            : extension revision 1
	VK_EXT_debug_report                    : extension revision 9
	VK_EXT_debug_utils                     : extension revision 2
	VK_EXT_direct_mode_display             : extension revision 1
	VK_EXT_display_surface_counter         : extension revision 1
	VK_KHR_device_group_creation           : extension revision 1
	VK_KHR_display                         : extension revision 21
	VK_KHR_external_fence_capabilities     : extension revision 1
	VK_KHR_external_memory_capabilities    : extension revision 1
	VK_KHR_external_semaphore_capabilities : extension revision 1
	VK_KHR_get_display_properties2         : extension revision 1
	VK_KHR_get_physical_device_properties2 : extension revision 1
	VK_KHR_get_surface_capabilities2       : extension revision 1
	VK_KHR_portability_enumeration         : extension revision 1
	VK_KHR_surface                         : extension revision 25
	VK_KHR_surface_protected_capabilities  : extension revision 1
	VK_KHR_wayland_surface                 : extension revision 6
	VK_KHR_xcb_surface                     : extension revision 6
	VK_KHR_xlib_surface                    : extension revision 6

Layers: count = 3
=================
VK_LAYER_MESA_device_select (Linux device selection layer) Vulkan version 1.3.211, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id = 0 (llvmpipe (LLVM 14.0.5, 256 bits))
		Layer-Device Extensions: count = 0

VK_LAYER_MESA_overlay (Mesa Overlay layer) Vulkan version 1.3.211, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id = 0 (llvmpipe (LLVM 14.0.5, 256 bits))
		Layer-Device Extensions: count = 0

VK_LAYER_NV_optimus (NVIDIA Optimus layer) Vulkan version 1.3.194, layer version 1:
	Layer Extensions: count = 0
	Devices: count = 1
		GPU id = 0 (llvmpipe (LLVM 14.0.5, 256 bits))
		Layer-Device Extensions: count = 0

Presentable Surfaces:
=====================
GPU id : 0 (llvmpipe (LLVM 14.0.5, 256 bits)):
	Surface types: count = 2
		VK_KHR_xcb_surface
		VK_KHR_xlib_surface
	Formats: count = 0
	Present Modes: count = 4
		PRESENT_MODE_IMMEDIATE_KHR
		PRESENT_MODE_MAILBOX_KHR
		PRESENT_MODE_FIFO_KHR
[... 842 lines cut ...]
	scalarBlockLayout                                  = true
	imagelessFramebuffer                               = true
	uniformBufferStandardLayout                        = true
	shaderSubgroupExtendedTypes                        = true
	separateDepthStencilLayouts                        = true
	hostQueryReset                                     = true
	timelineSemaphore                                  = true
	bufferDeviceAddress                                = true
	bufferDeviceAddressCaptureReplay                   = false
	bufferDeviceAddressMultiDevice                     = false
	vulkanMemoryModel                                  = false
	vulkanMemoryModelDeviceScope                       = false
	vulkanMemoryModelAvailabilityVisibilityChains      = false
	shaderOutputViewportIndex                          = true
	shaderOutputLayer                                  = true
	subgroupBroadcastDynamicId                         = true

VkPhysicalDeviceVulkanMemoryModelFeatures:
------------------------------------------
	vulkanMemoryModel                             = false
	vulkanMemoryModelDeviceScope                  = false
	vulkanMemoryModelAvailabilityVisibilityChains = false
```

<details>

DSchroer, 2023-02-15:
Looking deeper it seems like the next release of `bevy` will have much improved support for older GPU backends. I will try to update as soon as a stable version of it is available. 

https://github.com/bevyengine/bevy/issues/3191

setop, 2023-02-15:
On my debian 11, with an Intel HD 620, I had to [do](https://askubuntu.com/a/1098490) `sudo apt install mesa-vulkan-drivers`

closed 2023-02-16 by DSchroer