DSchroer/dslcad #12: Allow object style properties to access computed values

ephetic, 2023-02-15:
e.g. 
```
var table = 
    cube(x=10, y=10, z=10) 
    ->shape translate(x=1, y=2, z=3);
var vase = 
    cylinder(radius=3, height=3) 
    ->shape translate(x=table.center.x, y=table.center.y, z=table.center.z + table.z/2);
```

DSchroer, 2023-02-15:
Out of curiosity, what properties did you have in mind? Off the top of my head I can think of:
- center
- area
- volume

ephetic, 2023-02-15:
Honestly, I'd only thought of center, as it is another way I could have come at the centering (#14) for relative boolean geometry, but perhaps `transform` and/or `scale` would also be good.  I'm thinking that if we have property access, it makes it easy to define subordinate parts relative to a parent geometry (e.g. some sort of `->shape relative_to(parent=parent)` helper based on being able to extract the parent properties).

DSchroer, 2023-02-24:
Hey. Just a heads up that I am making some progress on this. Got a PR upstream to expose this stuff to the software and then once that is merged I can start to add the parameters. 

https://github.com/bschwind/opencascade-rs/pull/20

DSchroer, 2023-03-02:
Ive added the `x.center` property in the code. One thing I want to do is setup a `x ->shape center()` function. How would you use it? The options I see are 
```
->shape center(x=true, y=true)
->shape center(no_z=true)
```
I think it makes sense to have an axis control but I don't have a good signature yet.

ephetic, 2023-03-03:
I'd think `center(x=true, y=true, z=true)` where they all default to true.

DSchroer, 2023-03-03:
Ive though of that but it really doesn't make much sense. Boolean arguments are inherently false by default. One thing I thought of was maybe having a shorthand for boolean arguments so you could do this:
```
center(x,y)
```
which is the same as :
```
center(x=true, y=true)
```
Then having the unspecified value default to false still works and it would center x and y without z. However it makes `center()` kind of useless.

ephetic, 2023-03-03:
I thought the default was whatever you set it as in the file.ds var.

DSchroer, 2023-03-03:
Fair enough. Ive added:
```
- `center(shape=shape, x=[bool], y=[bool], z=[bool])` center a shape
```
The `x,y,z` arguments default to true. 
Should be available in pre-releases after `0.0.2.pre.4`

DSchroer, 2023-03-03:
Completed with https://github.com/DSchroer/dslcad/releases/tag/v0.0.2-pre.4

closed 2023-03-03 by DSchroer