DSchroer/dslcad #14: Include center parameter on 3D primitives

ephetic, 2023-02-15:
I tend to think from the global origin when figuring out my intersections/differences, so if I had my druthers, I'd make center=true the default for primitives versus placing their local origin at the global origin.  
Thanks!

DSchroer, 2023-02-15:
I cant make it the default as that would be a breaking change. However I am happy to add `center` to the 3D primitives. 

Out of curiosity do you think from the global origin for all three axis? I find myself thinking that way for x and y but for z I am usually not centering it along the axis. Hence I have centering for 2D but not 3D yet.

ephetic, 2023-02-15:
Ooh, you're right -- I think I only center xy in my head. I'm torn between that and the consistency of centering all three.

DSchroer, 2023-02-15:
I will center all three. I cant really justify centering only x,y since it really isn't consistent. Best option is probably:
```
square(x=x, y=y, center=true) ->shape extrude(z=z)
```
With this change you will still need
```
cube(x=x, y=y, z=z, center=true) ->shape translate(z=z/2)
```

ephetic, 2023-02-15:
Yeah, I agree.  Consistency is always best.

david-southern, 2023-02-20:
> cube(x=x, y=y, z=z, center=true) ->shape translate(z=z/2)

I was thinking about this as well.  Would this be an option:

```
cube(x=x, y=y, z=z, center = [ X_AXIS, Y_AXIS, Z_AXIS ] )
```

Where all of the axis constants are optional?  And perhaps for convenience:

```
cube(x=x, y=y, z=z, center = [ XY_AXIS ] )

cube(x=x, y=y, z=z, center = [ ALL_AXIS ] )
```

DSchroer, 2023-02-20:
Im starting to lean away from the center parameter actually. I know its convenient and we can leave it for these kinds of primitives. How do you all feel about having a center function instead:
```
cube() ->shape center(x=true)
``` 
I feel like this is better since you could center more complex geometry. It would come with computed values as mentioned in #12.

ephetic, 2023-02-21:
That would work for me, especially if it works for arbitrary geometry.

As an aside, how would you feel making `shape` the default argument for `->`, or perhaps the first declared parameter for any function/file.

DSchroer, 2023-02-21:
I think we are aligned on centering. For the pipe discussion I created a new thread and put my thoughts here: 
https://github.com/DSchroer/dslcad/discussions/18

DSchroer, 2023-03-03:
Fixed as of https://github.com/DSchroer/dslcad/releases/tag/v0.0.2-pre.4

DSchroer, 2023-03-03:
Using the `center()` function.

closed 2023-03-03 by DSchroer