DSchroer/dslcad #15: Arithmetic operators do not appear to respect order of operations

david-southern, 2023-02-20:
First of, let me say that I love this project!  It seems much more expressive than OpenSCAD, which I also love. :)

In tinkering with a basic model, I found that:

```
var baseHeight = 10;
var offset = 2;
translate(z = baseHeight / 2 + offset)
```
Results in a translation of 2.5 (i.e. 10 / (2 + 2) ), not 7, as I would expect

I then tried this:
```
translate(z = (baseHeight / 2) + offset)
```
And that does result in a translation of 7

DSchroer, 2023-02-20:
Makes sense. I think this can be classified as a bug and we can fix it up. It might break some existing code so lets do this sooner than later.

DSchroer, 2023-03-02:
This is now fixed in the code. Will cut a pre-release soon.

DSchroer, 2023-03-03:
Fixed as of https://github.com/DSchroer/dslcad/releases/tag/v0.0.2-pre.4

closed 2023-03-03 by DSchroer