DSchroer/dslcad #21: Shared library support

DSchroer, 2023-03-10:
Right now DSLCAD can call local scripts using `./path/to/file()`. This is a simple enough option but does not work if we want to efficiently share libraries. 

I am looking for a simple solution & syntax for how we can implemented shared libraries with the following properties:
1. Allow people to publish libs
2. Libs can reference other libs
3. All data of which libs are in use should be in the source.

gedw99, 2023-03-25:
this is really interesting.

these are plugins that people can write and use with dslcad ?

DSchroer, 2023-03-25:
Here is the general idea. Lets assume you are an expert in gears. So you build this cool dslcad script for making gears. It works like this:
```
./gear(pitch=...);
```
Now lets say I have a project and I want to use gears. There needs to be a way for us to share scripts so that my part can use your gear script. Which in turn I might want to share. So now we have:
```
./doms_robot(); 
   // using ./gear(...); 
```
So we end up with these nested parts. 
I would like to have something where I can say:
```
use url_to_gear_library as gears;

gears/gear(pitch=...);
```
And now we can all share the parts we are building.

DSchroer, 2023-12-05:
I have added the ability to load other file types or "resources" into the script. Right now its got `stl` and `ini` file support. The idea being that scripts and resources are treated equally. Then I add text rendering fonts will also be resources. 

With that the ability to pack and distribute bundles as a shared library is still an open question.

DSchroer, 2026-09-21:
Ended up going with `@` which indicates to do a module lookup for a folder called `modules`. Seems to work for now if you are manually building a library. Will need better packaging down the line but good for now

closed 2026-09-21 by DSchroer