DSchroer/dslcad #22: Lockup when opening a file on intel mac

gedw99, 2023-03-13:
first time it opened, and i opened the Table file example. All good.
i exported to stp file format. ll Good.

then i closed it...

second time it choked then i tried to open any sample file. Go the beachball of hell...

Its an old intel mac, so not even sure its worth fixing.. :)

AdapterInfo { name: "Intel Iris Graphics", vendor: 0, device: 0, device_type: IntegratedGpu, driver: "", driver_info: "", backend: Metal }

gedw99, 2023-03-25:
today i tried v0.0.2.

Its really nice system and the lang is great !!

The file opening issue seems to be non deterministic. 

seems to fail when the CPU starts grinding.

Is there a way to open a file from the cli so i can experiment more and try to make this reproducable ?

``` 
Usage: dslcad [OPTIONS] <SOURCE>

Arguments:
  <SOURCE>  Source path to load

Options:
  -o, --out <OUT>  Outdir [default: .]
  -h, --help       Print help
  -V, --version    Print version
```

gedw99, 2023-03-25:
i think it might be due to not releasing memory .. 

hence why it happens when i try to open a new project, as it has to release what it has.

DSchroer, 2023-03-25:
> Is there a way to open a file from the cli so i can experiment more and try to make this reproducable ?

Do you mean open the file in the editor from the CLI or just export the parts?

If you want to export the parts you can:
```
dslcad $PATH_TO_FILE -o $PATH_FOR_OUTPUT
```

I might be able to setup something to open the editor directly. Would this work for you?
```
dslcad open $PATH_TO_FILE
```

DSchroer, 2023-03-25:
Ill look into doing some quick memory profiling. Make sure that old renders are cleaned up properly.

gedw99, 2023-03-31:
Thanks…

dslcad open $PATH_TO_FILE 

Would be a nice feature …

DSchroer, 2024-01-14:
Hey. Do you mind trying with the new release? The CLI has been redone to make this easier plus new more stable build.