DSchroer/dslcad #23: Projections

gedw99, 2023-03-15:
Don’t know if anyone else is interested but I am an engineer and used CAD and video post software for years ..

anyway I am thinking it possible to output Elevations and Sections out of this 3d model.

the AST describing those can be the exact same format used for the 3D.

just a brain fart idea right now.

DSchroer, 2023-03-20:
Interesting idea. Could you elaborate on what the syntax and output would look like?

gedw99, 2023-03-25:
Just like a CAD tool, you can have views of the model from top, left, etc

the idea is to be able to tell it those view and then export them as 2D Projections.

the export format is up for discussion.  I would prefer something that is plugable.
- One is PDF
- another is the dslcad format itself because it can represent things in 2D from what i can see.

DSchroer, 2023-03-25:
>  is the dslcad format itself 

What is the DSLCAD format? I think I understand what you mean here but some more detail would help. 

So the way I see this is that I could develop two pieces that make this possible:
1. A new function called `project(x,y,z)` that would take an object and slice it along one of the three axis. This would take a 3D object and turn it into a 2D object like you had been asking for. 
2. A way to export 2D objects. Probably to SVG. The reason for SVG over PDF is that its a much easier format to work with. There are loads of tools that let you go from SVG to PDF (browsers, inkscape, photoshop, etc...) so I think that is reasonable. 

Do you think that would be enough to cover your needs?

gedw99, 2023-03-28:
You definitely got the gist of the whole idea. 

I see your point about svg . It’s more open and reusable.

gedw99, 2023-03-31:
in terms of how a user can store their preferences, some json file would be good. So a user can update them .

A fancy thing later is to update the svg whenever a section is changed. 

that’s tricky …

The sustainable way to do it is to export a data stream of geometry bounding boxes when ever a mutation occurs on the 3d model.

then each section preference is checked against the bounding box previous and the bounding box latest to get a diff. If it’s different you know the section needs to be re run.   

I don’t know how hard or quick it is to gen bounding boxes.

but that’s the high level conceptual idea of one way this could work to be push based.

The good thing about this is it can be run on different routines / processes and can scale out.

DSchroer, 2023-03-31:
> A fancy thing later is to update the svg whenever a section is effected by the 3d model change.

This feels in conflict with how DSLCAD works. Its essentially a compiler that creates 3D models. Typically compilers don't use CDCs and instead just re-build the output from scratch each time. I think for the foreseeable future we should avoid ALL state management. Instead lets focus on simply adding the projections and ability to produce SVGs in the end. If we keep this fast there will be no need for any state of CDC which simplifies the system significantly.

closed 2024-01-08 by DSchroer