DSchroer/dslcad #24: Try to use glam or nalgebra to replace OCC's transform

happydpc, 2023-03-31:
OCC's transform is too hard to use with rust, we can replace it with rust's math library, and cast back.

DSchroer, 2023-03-31:
I'm guessing you mean construct the transformation matrix in rust and then push that to a OCC transform? 

Sounds like a good plan. Feel free to submit a PR if you are interested in doing this.

DSchroer, 2023-03-31:
Out of curiosity, what are you attempting to achieve with this? Did you try something and run into limits or is this an improvement without an immediate need?

happydpc, 2023-03-31:
I want to use some exist rust repos to interact with occ.But the occ's gp_Trs or gp_GTrs is so hard to use, As we just need to pass a 4*3 or 4*4 array, just use a replaced type and cast to occ may ease the case.

DSchroer, 2023-03-31:
Makes sense. Lets start with getting the API you need in https://github.com/bschwind/opencascade-rs. After that we can use it here. What API for gp_Trs or gp_GTrs do you need in order to replace your transforms?

happydpc, 2023-03-31:
Maybe replace the types directly by glam::Affine3A and glam::Mat4, hope that user doesn't need to know occ's math types one day. 

I have tried truck repo before , it's still too young 😓.Using occ is the last choice.

bschwind, 2023-05-22:
Things are still majorly in flux, but for now I've started using `glam` in my [higher level OCC bindings](https://github.com/bschwind/opencascade-rs/tree/24223a4ab7c6696555cd8fa7a301406284d34f38/crates/opencascade)

I wouldn't recommend using it yet, but it's quite easy to integrate so there's no reason we can't use those more ergonomic types here.

DSchroer, 2023-05-22:
Makes sense. I am happy to accept a PR for that into DSLCAD. Right now my focus is on getting parity with the web editor and will move back to the language work after that.

closed 2026-09-21 by DSchroer