DSchroer/dslcad #25: build issue with linux, Ubuntu 22.04.3 LTS 64bit

minger0, 2023-10-08:
testuser@testuser-mycomputer:~/my/code/dslcad$ rustup run stable cargo --version
cargo 1.73.0 (9c4383fb5 2023-08-26)
testuser@testuser-mycomputer:~/my/code/dslcad$ rustup run stable cargo build
    Updating crates.io index
  Downloaded bevy_scene v0.10.1
  Downloaded ab_glyph v0.2.21
  Downloaded bevy_internal v0.10.1
  Downloaded bevy_macro_utils v0.10.1
  Downloaded block v0.1.6
  Downloaded bevy_log v0.10.1
  Downloaded cc v1.0.74
  Downloaded color_quant v1.1.0
  Downloaded cfg_aliases v0.1.1
  Downloaded bevy_render_macros v0.10.1
  Downloaded bevy_core v0.10.1
  Downloaded bevy_utils_proc_macros v0.10.1
  Downloaded beef v0.5.2
  Downloaded bevy_diagnostic v0.10.1
  Downloaded bevy_gilrs v0.10.1
  Downloaded bit-set v0.5.3
  Downloaded bevy_derive v0.10.1
  Downloaded adler v1.0.2
  Downloaded bevy_ecs_macros v0.10.1
  Downloaded bevy_encase_derive v0.10.1
  Downloaded bevy_a11y v0.10.1
  Downloaded bevy_input v0.10.1
  Downloaded logos v0.12.1
  Downloaded bevy_transform v0.10.1
  Downloaded ecolor v0.21.0
  Downloaded encase_derive v0.5.0
  Downloaded bevy_mikktspace v0.10.1
  Downloaded ab_glyph_rasterizer v0.1.8
  Downloaded anstyle-parse v0.2.1
  Downloaded bevy_gltf v0.10.1
  Downloaded bevy_animation v0.10.1
  Downloaded bevy_ptr v0.10.1
  Downloaded bevy_app v0.10.1
  Downloaded async-channel v1.9.0
  Downloaded anstyle v1.0.2
  Downloaded bevy_utils v0.10.1
  Downloaded concurrent-queue v2.2.0
  Downloaded bytemuck_derive v1.4.1
  Downloaded byteorder v1.4.3
  Downloaded com-rs v0.2.1
  Downloaded anstream v0.5.0
  Downloaded bevy_time v0.10.1
  Downloaded approx v0.5.1
  Downloaded async-executor v1.5.1
  Downloaded bevy_winit v0.10.1
  Downloaded parking v2.1.0
  Downloaded overload v0.1.1
  Downloaded num-rational v0.4.1
  Downloaded memoffset v0.6.5
  Downloaded accesskit_winit v0.12.4
  Downloaded cmake v0.1.49
  Downloaded bit-vec v0.6.3
  Downloaded bevy_math v0.10.1
  Downloaded clap_derive v4.4.2
  Downloaded bevy_points v0.2.1
  Downloaded subtle v2.5.0
  Downloaded pp-rs v0.2.1
  Downloaded bevy_tasks v0.10.1
  Downloaded bevy_ui v0.10.1
  Downloaded bevy_text v0.10.1
  Downloaded ahash v0.7.6
  Downloaded grid v0.10.0
  Downloaded bzip2 v0.4.4
  Downloaded bevy_asset v0.10.1
  Downloaded bevy_hierarchy v0.10.1
  Downloaded base64ct v1.6.0
[... 429 lines cut ...]
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_LIBDIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64-unknown-linux-gnu
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR_x86_64_unknown_linux_gnu
  cargo:rerun-if-env-changed=HOST_PKG_CONFIG_SYSROOT_DIR
  cargo:rerun-if-env-changed=PKG_CONFIG_SYSROOT_DIR

  --- stderr
  thread 'main' panicked at /home/testuser/.cargo/registry/src/index.crates.io-6f17d22bba15001f/alsa-sys-0.3.1/build.rs:13:18:
  `PKG_CONFIG_ALLOW_SYSTEM_CFLAGS="1" PKG_CONFIG_ALLOW_SYSTEM_LIBS="1" "pkg-config" "--libs" "--cflags" "alsa"` did not exit successfully: exit status: 1
  error: could not find system library 'alsa' required by the 'alsa-sys' crate

  --- stderr
  Package alsa was not found in the pkg-config search path.
  Perhaps you should add the directory containing `alsa.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'alsa' found

  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
warning: build failed, waiting for other jobs to finish...

---
any suggestion on how to proceed?

minger0, 2023-10-14:
after 
```
sudo apt install libasound2-dev libudev-dev
```
the [opencascade libs](https://techoverflow.net/2022/11/26/how-to-install-opencascade-on-ubuntu-22-04/),
```
sudo apt install -y xfonts-scalable libocct-data-exchange-dev libocct-draw-dev libocct-foundation-dev libocct-modeling-algorithms-dev libocct-modeling-data-dev libocct-ocaf-dev libocct-visualization-dev
```
were needed to be installed
```
$ rustup run stable cargo build
   Compiling zstd-sys v2.0.8+zstd.1.5.5
   Compiling zstd-safe v5.0.2+zstd.1.5.2
   Compiling opencascade-sys v0.2.0 (https://github.com/bschwind/opencascade-rs?rev=d9391e43bffca1356c1cf74cd15837f2ceb6286e#d9391e43)
error: failed to run custom build command for `opencascade-sys v0.2.0 (https://github.com/bschwind/opencascade-rs?rev=d9391e43bffca1356c1cf74cd15837f2ceb6286e#d9391e43)`

Caused by:
  process didn't exit successfully: `/home/testuser/my/code/dslcad/target/debug/build/opencascade-sys-50caed4f7197da8d/build-script-build` (exit status: 101)
  --- stdout
  CMAKE_TOOLCHAIN_FILE_x86_64-unknown-linux-gnu = None
  CMAKE_TOOLCHAIN_FILE_x86_64_unknown_linux_gnu = None
  HOST_CMAKE_TOOLCHAIN_FILE = None
  CMAKE_TOOLCHAIN_FILE = None
  CMAKE_GENERATOR_x86_64-unknown-linux-gnu = None
  CMAKE_GENERATOR_x86_64_unknown_linux_gnu = None
  HOST_CMAKE_GENERATOR = None
  CMAKE_GENERATOR = None
  CMAKE_PREFIX_PATH_x86_64-unknown-linux-gnu = None
  CMAKE_PREFIX_PATH_x86_64_unknown_linux_gnu = None
  HOST_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_x86_64-unknown-linux-gnu = None
  CMAKE_x86_64_unknown_linux_gnu = None
  HOST_CMAKE = None
  CMAKE = None
  running: cd "/home/testuser/my/code/dslcad/target/debug/build/opencascade-sys-54e19048fb829fa7/out/build" && CMAKE_PREFIX_PATH="" "cmake" "/home/testuser/.cargo/git/checkouts/opencascade-rs-56632466c8b8dee3/d9391e4/crates/opencascade-sys/OCCT" "-DCMAKE_INSTALL_PREFIX=/home/testuser/my/code/dslcad/target/debug/build/opencascade-sys-54e19048fb829fa7/out" "-DCMAKE_C_FLAGS= -ffunction-sections -fdata-sections -fPIC -m64" "-DCMAKE_C_COMPILER=/usr/bin/cc" "-DCMAKE_CXX_FLAGS= -ffunction-sections -fdata-sections -fPIC -m64" "-DCMAKE_CXX_COMPILER=/usr/bin/c++" "-DCMAKE_ASM_FLAGS= -ffunction-sections -fdata-sections -fPIC -m64" "-DCMAKE_ASM_COMPILER=/usr/bin/cc" "-DCMAKE_BUILD_TYPE=Debug"
  -- Configuring incomplete, errors occurred!
  See also "/home/testuser/my/code/dslcad/target/debug/build/opencascade-sys-54e19048fb829fa7/out/build/CMakeFiles/CMakeOutput.log".

  --- stderr
  CMake Error at CMakeLists.txt:3 (find_package):
    By not providing "FindOpenCASCADE.cmake" in CMAKE_MODULE_PATH this project
    has asked CMake to find a package configuration file provided by
    "OpenCASCADE", but CMake did not find one.

    Could not find a package configuration file provided by "OpenCASCADE" with
    any of the following names:

      OpenCASCADEConfig.cmake
      opencascade-config.cmake

    Add the installation prefix of "OpenCASCADE" to CMAKE_PREFIX_PATH or set
    "OpenCASCADE_DIR" to a directory containing one of the above files.  If
    "OpenCASCADE" provides a separate development package or SDK, be sure it
    has been installed.

  thread 'main' panicked at /home/testuser/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cmake-0.1.49/src/lib.rs:1104:5:

  command did not execute successfully, got: exit status: 1

  build script failed, must exit now
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
  thread 'main' panicked at /home/testuser/.cargo/git/checkouts/opencascade-rs-56632466c8b8dee3/d9391e4/crates/opencascade-sys/build.rs:92:23:
  Pre-installed OpenCASCADE library not found. You can use `builtin` feature if you do not want to install OCCT libraries system-wide.: Any { .. }
warning: build failed, waiting for other jobs to finish...
```
pls help me to get the build running

DSchroer, 2023-10-15:
Hey. Sorry about the late reply. Try compiling with `-F builtin-occt` its a flag that should build opencascade with the application. If not you need to clone the OCCT repo and manually build it with the following config:
```
cmake \
    -DBUILD_LIBRARY_TYPE=Static \
    -DBUILD_MODULE_ApplicationFramework=FALSE \
    -DBUILD_MODULE_Draw=FALSE \
    -DUSE_D3D=FALSE \
    -DUSE_DRACO=FALSE \
    -DUSE_EIGEN=FALSE \
    -DUSE_FFMPEG=FALSE \
    -DUSE_FREEIMAGE=FALSE \
    -DUSE_FREETYPE=FALSE \
    -DUSE_GLES2=FALSE \
    -DUSE_OPENGL=FALSE \
    -DUSE_OPENVR=FALSE \
    -DUSE_RAPIDJSON=FALSE \
    -DUSE_TBB=FALSE \
    -DUSE_TCL=FALSE \
    -DUSE_TK=FALSE \
    -DUSE_VTK=FALSE \
    -DUSE_XLIB=FALSE \
    -DBUILD_MODULE_DETools=FALSE
```
Its not a very smooth setup right now.

DSchroer, 2024-01-10:
I have improved the build process. There is now a justfile that acts as a build script to setup OCCT. Full build should work by running:
```
just build
```

It takes care of pre-building OCCT the first time and uses the correct env variables.

closed 2024-01-10 by DSchroer