DSchroer/dslcad #26: some initial problems - Windows build

lafar6502, 2024-01-13:
Hi, im checking the windows version from releases page (March 2023)
and have noticed some problems

1. dslcad --help doesnt display anything
2. when trying to load a table example from command line: dslcad.exe table.ds - nothing happens. no message, no gui
3. when opening table.ds example, a syntax error is reported
And the most important - documentation seems very limited.  
the tool seems promising and interesting, but without initial knowledge of it it's almost impossible to get started and achieve anything useful

DSchroer, 2024-01-13:
Hey. Thanks for reaching out. I am working on a new release in the next week or so. Should improve the windows version quite a bit.

DSchroer, 2024-01-14:
The release is ready. Do you mind giving it another try?

https://github.com/DSchroer/dslcad/releases/edit/v0.0.3

I have also updated the web editor and plan on working on improving the docs now.

lafar6502, 2024-01-14:
Thank you! 
And yes, looks like things improved a bit - table example is working now, there's command line help....
Next question - i see it exports 3mf format now. This format can store color/material information (at least ive read that it does), but does dslcad language allow to define materials and use them for models?

lafar6502, 2024-01-14:
Additional questions:

how do we parametrize a script? For example, i have table.ts script, but when calling dslcad to render it to 3mf file, i'd like to specify some initial variables, like width, height and depth of the table. Does command line allow that?

And what is the underlying geometry engine in dslcad? Is it based on open cascade?

DSchroer, 2024-01-14:
> but does dslcad language allow to define materials and use them for models?

Not yet but it is definitely in the plan. `0.0.3` put the groundwork together needed for more complex multi part designs and materials can now be added in a way that makes sense. 

> how do we parametrize a script?

Right now the easiest way is to use another script or a `*.ini` file. If its something that you are looking for in CLI commands it should be easy to add. 

## Ini Method

For the ini method. Assume I have a file called `params.ini`:
```txt
length=10
width=50
```

Then in my script I can do:
```
var params = ./params.ini();

cube(x=params.width, y=params.height);
```

## Parent Script Method

Say I have a script (`cube.ds`) like so:
```
var width = 10;
var height = 50;

cube(x=width, y=height);
```

I can have another script that calls it like this:

```
./cube.ds(width=50, height=100);
```

DSchroer, 2024-01-14:
> And what is the underlying geometry engine in dslcad? Is it based on open cascade?

Yes. DSLCAD uses OpenCascade under the hood.

lafar6502, 2024-01-14:
Thanks. My use case is 3D visualization of configurable products, like blinds, roofs, awnings - i thought it can be easier to do with a parametric CAD tool instead of general-purpose 3D scene description. Based on the configuration (size, colors, various component options) dslcad would generate a 3D product representation and convert that to 3D file like 3mf. Then the file could be displayed in a web browser. And after every update of product configuration the 3D file would be recreated to show the updated look.
Do you think dslcad would be fast enough for such visualization?

DSchroer, 2024-01-14:
It really depends on how complex the model is. 
OpenCascade is really the bottleneck in most cases. If you are doing simple operations it should be reasonably fast.
For a well designed model I think a parametric configurator should work fine. 

BTW the WASM build runs with the same CLI interface. So you could even have an online hosted configurator that runs client side. As long as you build the right experience with it.

DSchroer, 2024-01-14:
Currently it takes 2.3 seconds to run ALL the examples.

DSchroer, 2024-01-15:
Marking this issue as closed since Windows builds are working again.

closed 2024-01-15 by DSchroer