DSchroer/dslcad #30: build instructions for Windows

lafar6502, 2024-03-08:
hi, are there any instructions how to build the source code in Windows (from scratch)
I have Rust and Just installed but 'just build' doesnt seem to just work
The error says shell is missing but i'm pretty discouraged by the need to dig into just tool now and figure out what is the problem. Probably simple but just no, i dont want to start learning new foreign tool now, Rust is foreign enough to me. If you have some simple solution to get it to compile i'll be very grateful.
```
D:\devs2\dslcad>just build
error: Backtick could not be run because just could not find the shell:
program not found
 ——▶ justfile:2:38
  │
2 │ export CMAKE_BUILD_PARALLEL_LEVEL := `nproc --all`
  │                                      ^^^^^^^^^^^^^

```

DSchroer, 2024-03-08:
Hey. Can you try running `cargo build` directly?

lafar6502, 2024-03-08:
Yes, almost successful, i think
but

```
Compiling dslcad-viewer v0.0.3 (D:\devs2\dslcad\crates\dslcad_viewer)
error: environment variable `HOME` not defined at compile time
  --> crates\dslcad_viewer\src\settings.rs:49:37
   |
49 |         let config_file = Path::new(env!("HOME"))
   |                                     ^^^^^^^^^^^^
   |
   = help: use `std::env::var("HOME")` to read the variable at run time
   = note: this error originates in the macro `env` (in Nightly builds, run with -Z macro-backtrace for more info)
```

what's supposed to be in HOME env variable?

DSchroer, 2024-03-08:
Interesting. Yeah HOME is the config dir but I guess it does not exist for Windows. I just pushed a fix using the `simple-home-dir` crate which should work on Windows. Pull latest and try now.

lafar6502, 2024-03-09:
yep, your change fixes the problem, just needed to apply it in next function too

```
 crates/dslcad_viewer/src/settings.rs | 5 +++--
 1 file changed, 3 insertions(+), 2 deletions(-)

diff --git a/crates/dslcad_viewer/src/settings.rs b/crates/dslcad_viewer/src/settings.rs
index 215760d..df246fb 100644
--- a/crates/dslcad_viewer/src/settings.rs
+++ b/crates/dslcad_viewer/src/settings.rs
@@ -66,8 +66,9 @@ impl FileStore {
             writeln!(output, "{key}={value}").expect("failed to write value");
         }
 
-        let config_dir = Path::new(env!("HOME")).join(".config").join("dslcad");
-
+        let config_dir = Path::new(&home_dir().expect("unable to locate home directory"))
+            .join(".config")
+            .join("dslcad");
         fs::create_dir_all(&config_dir).expect("failed to create config dir");
         fs::write(config_dir.join("config.ini"), output).expect("failed to save config");
     }

```

DSchroer, 2024-03-09:
Nice. Added that fix to main as well.

DSchroer, 2024-03-09:
Resolving as fixed.

closed 2024-03-09 by DSchroer