DSchroer/dslcad #31: DSLcad language - control statements

lafar6502, 2024-07-07:
Hi, in the documentation of dslcad language syntax i dont see any information about
* if statement
* for loop (or any other loop)
* functions (definition of own functions/subroutines)
Are these elements present in the language?

DSchroer, 2024-07-08:
Hey. You are correct they are missing from the concepts page. They could use more documentation. 

- `if` you can find in the syntax reference page: https://dslcad.com/reference/
- there are no for loops but you can use `map` and `reduce` as well as `range` to iterate over parts/values. 
- `functions` are present but undocumented for the time being. They are similar to scopes but you use them as follows: `var f = func { 5; };` and then you can invoke it later. You can also use separate files and invoke them as if they were functions.

DSchroer, 2025-01-30:
Updated the docs to include control flow info: https://dslcad.com/concepts/#control-flow

closed 2025-01-30 by DSchroer